indilib/pyindi-client #1: Failed to install/build 0.2.7 version

artak-kirakosyan, 2021-07-25:
I am failing to install 0.2.7 version on my Debian on AWS(Linux 4.19.0-17-cloud-amd64 #1 SMP Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux)

What I did is first downloaded, built and installed indi core(it works fine, version 1.9.0)

Then I tried installing pyindi-client(of course after installing the dependencies):
The output is way too long to put here so I am attaching it in a [file](https://github.com/indilib/pyindi-client/files/6874004/output.txt)

Below you can see the versions of dependency packages: 
```
libcfitsio-dev is already the newest version (3.450-3).
libindi-dev is already the newest version (1.7.5+dfsg-1).
libnova-dev is already the newest version (0.16-4).
python3-setuptools is already the newest version (40.8.0-1).
python3-dev is already the newest version (3.7.3-1).
swig is already the newest version (3.0.12-2).
```
I am using python3.7.3

Can anyone help me understand what's wrong here and how to fix it?

P.S. 0.2.6 version works fine.

knro, 2021-07-25:
libindi is too old, you need to install libindi v1.9.1

artak-kirakosyan, 2021-07-25:
I might have missed something, but the latest stable version of Debian is the buster(which is installed on my VM) and the latest available version of libindi for it is 1.7.5.
Am I missing something or is there a way to forcefully install libindi v1.9.1 on older version of Debian?

knro, 2021-07-25:
Looks like debian is way behind in packaging INDI. You need to compile it from Github.

artak-kirakosyan, 2021-07-25:
thanks, it worked.
I removed the libindi with apt and built from sources, and now it works fine.

closed 2021-07-25 by artak-kirakosyan