indilib/pyindi-client #2: Getting started example does not work

clkotnik, 2021-08-22:
When I run the getting started example program it failed on line 83 with:

> TypeError: 'SwigPyObject' object is not iterable

Complete output here:
[output.txt](https://github.com/indilib/pyindi-client/files/7027953/output.txt)

My environment is:
ubuntu 20.04
Python 3.8.10
indi-full 1.9.1 package from ppa:mutlaqja/ppa
pyindi-client                 0.2.7 installed with pip

best regards,
Cliff

aaronwmorris, 2021-08-23:
PyPI does not have the latest code for pyindi-client.  You will need to install directly from git

`pip install git+https://github.com/indilib/pyindi-client.git@68803a1#egg=pyindi-client`

fockez, 2021-09-18:
install new code is no help

gnthibault, 2021-09-30:
Same issue here

aaronwmorris, 2021-10-01:
> Also, before, it was possible to get acces to attribute "name" on each property, now here is what I get from the returned value from findWIdgetByName:
> 
> ```
> test = pv.findWidgetByName("CONNECT")
> test.name
> Traceback (most recent call last):
>   File "/home/user/venv/lib/python3.8/site-packages/IPython/core/interactiveshell.py", line 3444, in run_code
>     exec(code_obj, self.user_global_ns, self.user_ns)
>   File "<ipython-input-5-51046e3b0758>", line 1, in <module>
>     test.name
> AttributeError: 'SwigPyObject' object has no attribute 'name'
> ```

I think direct access to some of the attributes are no longer available.  I had to convert to using functions to fetch the attributes, such as getName() instead of accessing .name

gnthibault, 2021-10-01:
Based on https://github.com/indilib/indi/blob/723ae45593d204388cec337396be4c3b8300255c/libs/indibase/property/indipropertyview.h#L159 I was curious wether @pawel-soja had an idea about what is going on ? might create an issue in main indi repo

artak-kirakosyan, 2021-11-06:
> PyPI does not have the latest code for pyindi-client. You will need to install directly from git
> 
> `pip install git+https://github.com/indilib/pyindi-client.git@68803a1#egg=pyindi-client`

Installing from this link solved the issue for me, however a lot of object types have changed, for example, 'ITextVectorProperty' became 'PropertyViewText' .

artak-kirakosyan, 2021-11-06:
An alternative to git install is using
 ```pip install pyindi-client==0.2.7b0```
 
I believe we just need a new release for PyPi so that people can get the fresh fixes from PyPi directly.
@knro what do you think?

knro, 2021-11-06:
I agree, we really need someone more versed in Python to take of of pyindi-client, I'm already busy with the C++ INDI. Any volunteers who can manage this?

artak-kirakosyan, 2021-11-06:
I wouldn't say I am an advanced Python engineer, but I am using this in my everyday routine so I would really like to make it a good library.

I will start researching about how the PyPi releases work and update here.

Are there any better channels for communication or this Issue is good enough?

knro, 2021-11-06:
This issue is good. I asked Jean-Luc to transfer PyPi over for a new maintainer.

knro, 2021-11-06:
@artak-kirakosyan If you have a PyPi username, please let me know so I add you as maintainer.

artak-kirakosyan, 2021-11-06:
@knro  here it is: artak-kirakosyan

knro, 2021-11-06:
I sent you an invitation, please check.

artak-kirakosyan, 2021-11-06:
I haven't received anything, could you please double check if you sent it to this profile? https://pypi.org/user/artak-kirakosyan
You can also try to find me via my email: artakkirakosyan96@gmail.com

artak-kirakosyan, 2021-11-22:
@clkotnik @aaronwmorris @fockez @gnthibault I uploaded the latest version to PyPi.
Please install the 0.2.8 version and let me know if the issue is not fixed

aaronwmorris, 2021-11-22:
Worked for me.

artak-kirakosyan, 2021-12-20:
@knro please close this issue or give me access to do it.
Looks like updating the version fixed the problem.

knro, 2021-12-21:
@artak-kirakosyan You're the repo maintainer now!

closed 2021-12-21 by artak-kirakosyan

abecadel, 2022-01-10:
@knro  I would like to help maintaining this repo as well

knro, 2022-01-10:
Sure @abecadel , you have access now. Please coordinate with @artak-kirakosyan