indilib/pyindi-client #4: TypeError: 'SwigPyObject' object is not iterable

gerben-b, 2021-12-31:
**Describe the bug**
I installed indi and indi-3rdparty and pyindi-client from the latest source (indi 1.9.3) according to the respective README files and ran the example in README.rst in the pyindi-client repository. I ran the server as "indiserver indi_simulator_telescope indi_simulator_ccd"
I got this result:
_2021-12-30 18:05:46,465 creating an instance of IndiClient
Connecting and waiting 1 sec
INDI::BaseClient::connectServer: creating new connection...
2021-12-30 18:05:46,465 Server connected (localhost:7624)
INDI::BaseClient::connectServer: Already connected.
2021-12-30 18:05:46,466 new device CCD Simulator
2021-12-30 18:05:46,466 new property CONNECTION for device CCD Simulator
2021-12-30 18:05:46,466 new property DRIVER_INFO for device CCD Simulator
2021-12-30 18:05:46,467 new property POLLING_PERIOD for device CCD Simulator
2021-12-30 18:05:46,467 new property DEBUG for device CCD Simulator
2021-12-30 18:05:46,467 new property CONFIG_PROCESS for device CCD Simulator
2021-12-30 18:05:46,467 new property ACTIVE_DEVICES for device CCD Simulator
2021-12-30 18:05:46,467 new property SIMULATOR_SETTINGS for device CCD Simulator
2021-12-30 18:05:46,467 new device Telescope Simulator
2021-12-30 18:05:46,467 new property CONNECTION for device Telescope Simulator
2021-12-30 18:05:46,467 new property DRIVER_INFO for device Telescope Simulator
2021-12-30 18:05:46,467 new property POLLING_PERIOD for device Telescope Simulator
2021-12-30 18:05:46,467 new property DEBUG for device Telescope Simulator
2021-12-30 18:05:46,468 new property CONFIG_PROCESS for device Telescope Simulator
2021-12-30 18:05:46,468 new property CONNECTION_MODE for device Telescope Simulator
2021-12-30 18:05:46,468 new property DEVICE_PORT for device Telescope Simulator
2021-12-30 18:05:46,468 new property DEVICE_BAUD_RATE for device Telescope Simulator
2021-12-30 18:05:46,468 new property DEVICE_AUTO_SEARCH for device Telescope Simulator
2021-12-30 18:05:46,468 new property DEVICE_PORT_SCAN for device Telescope Simulator
2021-12-30 18:05:46,468 new property ACTIVE_DEVICES for device Telescope Simulator
2021-12-30 18:05:46,468 new property DOME_POLICY for device Telescope Simulator
2021-12-30 18:05:46,468 new property TELESCOPE_INFO for device Telescope Simulator
2021-12-30 18:05:46,468 new property SCOPE_CONFIG_NAME for device Telescope Simulator
2021-12-30 18:05:46,468 new property MOUNT_TYPE for device Telescope Simulator
2021-12-30 18:05:46,468 new property SIM_PIER_SIDE for device Telescope Simulator
2021-12-30 18:05:46,468 new property MOUNT_MODEL for device Telescope Simulator
2021-12-30 18:05:46,468 new property MOUNT_AXES for device Telescope Simulator
2021-12-30 18:05:46,468 new property FLIP_HA for device Telescope Simulator
2021-12-30 18:05:46,469 new property EQUATORIAL_PE for device CCD Simulator
2021-12-30 18:05:46,469 new property SIM_FOCUSING for device CCD Simulator
2021-12-30 18:05:46,469 new property SIMULATE_BAYER for device CCD Simulator
2021-12-30 18:05:46,469 new property CCD_SIMULATE_CRASH for device CCD Simulator
List of devices
CCD Simulator
Telescope Simulator
List of Device Properties
-- CCD Simulator
Traceback (most recent call last):
File "/home/gerben/indi.py", line 82, in
for p in lp:
TypeError: 'SwigPyObject' object is not iterable
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.To Reproduce_

**Exact steps to reproduce the behavior.**

Start the server as indicated above
Run the example as indicated above
See error
**Expected behavior**
A result without the error message.

**Log Files**
See the above output

artak-kirakosyan, 2022-01-03:
which version of pyindi are you using?
if it is not the 0.2.8, please install it and try again.

gnthibault, 2022-01-07:
I also have this problem:

```
 /usr/bin/python3 /usr/bin/indi-mqtt.py
indi-mqtt: Using configuration from /etc/indi-mqtt.conf
indi-mqtt: Starting in auto refresh mode in every 10 seconds.
indi-mqtt: Creating an instance of INDI client
INDI::BaseClient::connectServer: creating new connection...
indi-mqtt: Connected to MQTT server 192.168.0.48:1883
indi-mqtt: Subscribed to observatory/poll topic at MQTT server 192.168.0.48:1883
INDI::BaseClient::connectServer: Already connected.
indi-mqtt: Connected to INDI server localhost:7624
--- Logging error ---
Traceback (most recent call last):
  File "/usr/bin/indi-mqtt.py", line 461, in <module>
    observatory_json = getJSON(devices)
  File "/usr/bin/indi-mqtt.py", line 265, in getJSON
    for property in properties:
TypeError: 'SwigPyObject' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1083, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 927, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 663, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/bin/indi-mqtt.py", line 471, in <module>
    logger.error("Unexpected error timed polling INDI server", err)
Message: 'Unexpected error timed polling INDI server'
Arguments: (TypeError("'SwigPyObject' object is not iterable"),)
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.
--- Logging error ---
Traceback (most recent call last):
  File "/usr/bin/indi-mqtt.py", line 461, in <module>
    observatory_json = getJSON(devices)
  File "/usr/bin/indi-mqtt.py", line 265, in getJSON
    for property in properties:
TypeError: 'SwigPyObject' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1083, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 927, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 663, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/bin/indi-mqtt.py", line 471, in <module>
    logger.error("Unexpected error timed polling INDI server", err)
Message: 'Unexpected error timed polling INDI server'
Arguments: (TypeError("'SwigPyObject' object is not iterable"),)
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.
--- Logging error ---
Traceback (most recent call last):
  File "/usr/bin/indi-mqtt.py", line 461, in <module>
    observatory_json = getJSON(devices)
  File "/usr/bin/indi-mqtt.py", line 265, in getJSON
    for property in properties:
[... 64 lines cut ...]
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/bin/indi-mqtt.py", line 471, in <module>
    logger.error("Unexpected error timed polling INDI server", err)
Message: 'Unexpected error timed polling INDI server'
Arguments: (TypeError("'SwigPyObject' object is not iterable"),)
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.
^CTraceback (most recent call last):
  File "/usr/bin/indi-mqtt.py", line 450, in <module>
    time.sleep(1)
KeyboardInterrupt
```

pip freeze | grep pyindi
pyindi-client==0.2.8

indiserver --version
2022-01-07T20:52:01: startup: indiserver --version 
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 1.9.3
Code v1.9.3-39-gce7416338. Protocol 1.7.

also referenced in https://github.com/rkaczorek/indi-mqtt/issues/11

abecadel, 2022-01-13:
I was able to reproduce this error in test https://github.com/indilib/pyindi-client/runs/4797923354
Will look into this

gerben-b, 2022-01-20:
In the mean time, I built an xml parser in python to communicate with the indiserver so I don't need this anymore. So, no need to hurry ;-)

abecadel, 2022-01-23:
reopenning as it's still an issue to fix

abecadel, 2022-01-23:
It appears that the bug was introduced after indi v1.9.0 in https://github.com/indilib/indi/issues/1449 and https://github.com/indilib/indi/pull/1453 - property API has changed

artak-kirakosyan, 2022-01-23:
@abecadel you are right.
Looks like this problem is present for getProperties method only.
The rest of iterators are fine.

Do you have any experience with SWIG interface files?
I haven't done anything with them so far, if you have been already exposed to it, I would appreciate your help on fixing this.

knro, 2022-02-16:
Any update on this issue? what can be done to fix it?

abecadel, 2022-03-05:
> Any update on this issue? what can be done to fix it?

Unfortunately I was unable to fix it myself. 
The problem is a new version of properties in libindi is incompatible with current  indiclientpython.i, so it needs to be changed accordingly
> It appears that the bug was introduced after indi v1.9.0 in https://github.com/indilib/indi/issues/1449 and https://github.com/indilib/indi/pull/1453 - property API has changed