indilib/pyindi-client #21: Sevfault on property.getName()

aaronwmorris, 2022-08-26:
I have not found the exact combination of libraries or system that causes this, but I have a system that is generating a segfault when calling getName() on a property via pyindi-client.

System:  Raspberry Pi 3
Platform: armhf
OS:  Raspbian 10
Python: 3.7.3
indi:  1.9.7 (latest via astroberry repo)
pyindi-client:  1.9.1

```
$ gdb /home/aaron/git/indi-allsky/virtualenv/indi-allsky/bin/python3 core
GNU gdb (Raspbian 8.2.1-2) 8.2.1
<snip>

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/aaron/git/indi-allsky/virtualenv/indi-allsky/bin/python3...(no debugging symbols found)...done.
[New LWP 4103]
[New LWP 4106]
[New LWP 4107]
[New LWP 4109]
[New LWP 4108]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `python3 ./allsky.py run'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x6ef7b2f2 in INDI::Property::getName() const ()
   from /home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/_PyIndi.cpython-37m-arm-linux-gnueabihf.so
[Current thread is 1 (Thread 0x76f35b40 (LWP 4103))]
(gdb) where
#0  0x6ef7b2f2 in INDI::Property::getName() const ()
    at /home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/_PyIndi.cpython-37m-arm-linux-gnueabihf.so
#1  0x6ef0ba4c in _wrap_Property_getName.lto_priv.779 ()
    at /home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/_PyIndi.cpython-37m-arm-linux-gnueabihf.so
#2  0x001bf764 in _PyMethodDef_RawFastCallKeywords ()
#3  0x0011e880 in  ()
```

aaronwmorris, 2022-08-26:
Yes, it is just trying to run getName() on each property returned by device.getProperties()

https://github.com/aaronwmorris/indi-allsky/blob/master/indi_allsky/camera/indi.py#L293

I cannot prove this, but I suspect this might be a problem with the indi as distributed by astroberry.  I have other all sky cameras running slight variations of the versions, but none have this problem.  I had a report from someone on facebook that seemed to be having the segfault issue and I just happen to run into the same thing on one of my test cameras.

The hardware does not seem to matter, as I had the same segfault with a ZWO and PlayerOne device connected.

In the meantime, I just removed the getName() calls for devices as it was not really necessary for my software to function.

I am still investigating.

aaronwmorris, 2022-08-26:
And for further clarification, it only seems to happen when calling getName() on device properties.  Control properties seem to be unaffected.

abecadel, 2022-11-25:
Can you try that again with newest version? what is the system? what are the versions of pyindi & indi? how do you trigger it?

aaronwmorris, 2022-11-27:
I cannot recreate this segfault anymore.  I guess I will close this for now.  Thanks for looking into it!

closed 2022-11-27 by aaronwmorris