indilib/pyindi-client #25: Howto install Pyindi-client on Ubuntu

f6bvp, 2022-11-01:
Freshly installed Ubuntu 22.04 on a Raspberry Pi4b is working fine.
pyindi-client install needs pre requisit  :
sudo apt install python3-setuptools
sudo apt install python-dev-is-python3 libindi-dev swig
sudo apt install python3-pip
sudo apt-get install libcfitsio-dev
sudo apt-get install libnova-dev
sudo pip install pyindi-client
Succes !

All this sequence is not clearly given on different sites.
For example on : 
https://pypi.org/project/pyindi-client/
https://github.com/indilib/pyindi-client

Bernard
President dimension Parabole
http://radiotelescope-lavillette.fr

f6bvp, 2022-11-01:
This page should be updated for Ubuntu 22.04
https://indilib.org/support/tutorials/166-installing-and-using-the-python-pyndi-client-on-raspberry-pi.html

knro, 2022-11-02:
Yes. I asked for volunteers to maintain the pyindi-client documentation on INDI website and elsewhere. If you're up for it, please let me know!

Italianmoose, 2023-07-12:
Confirmed this worked on an RPi4 (using the Astroberry image), I couldn't get it to work on a Windows Subsystem for Linux using Ubuntu

f6bvp, 2023-08-04:
For Ubuntu 23.04 the solution I found in order to install PyIndi-client was to download sources from 

https://github.com/indilib/pyindi-client.git

Enter a Linux console and run command :
git clone https://github.com/indilib/pyindi-client.git
Then cd pyindi-client

Make compilation according to README executing command :

python setup.py install
did not find python !
I had to create a symbolic link :
sudo ln -s /usr/bin/python3.11 /usr/bin/python

Despite a lot of warnings pyindi-client PyIndi is now available in Python applications by 
import pyindi 

I hope this help.

knro, 2025-05-04:
We released an updated version, please check.