indilib/pyindi-client #26: Segfault with INDI 1.9.9

aaronwmorris, 2022-11-23:
Some change in the indi core library is causing a segfault or other vague errors with pyindi-client (HEAD).  I believe whatever is causing the segfault was added after Nov 6th.

pyindi-client is completely non-functional with indi 1.9.9

knro, 2022-11-24:
@pawel-soja Can you please check?

@aaronwmorris Can you share how we can reproduce this issue exactly step by step?

pawel-soja, 2022-11-24:
@aaronwmorris do you have any logs? Have you recompiled all dependent libraries (e.g. indi-3rdparty)?
Please provide additional information.

abecadel, 2022-11-25:
I cannot see indi beeing tagged at 1.9.9 yet
1.9.8 seems to work (with the simple test we've got atm) https://github.com/indilib/pyindi-client/actions/runs/3238312112/jobs/5306396117
building & testing against the current master atm https://github.com/indilib/pyindi-client/actions/runs/3544738039 

@aaronwmorris what exactly caused the segfault, please add a test for it or at least an instrution on how to reproduce it

BTW: it might be a good idea to create more tests to cover more area!

aaronwmorris, 2022-11-26:
The indi Ubuntu PPA repo has already released packages with the 1.9.9 release.  I have also tested by compiling from source.

I finally had some time tonight to write some basic code and there seems to be a problem with setting properties.  None of the current script examples that set properties in the repo are working.

If you already have pyindi-client installed, you will have to run the following to ensure you get the latest version from git.

```
pip3 install --no-binary :all: --upgrade 'git+https://github.com/indilib/pyindi-client.git@ffd939b#egg=pyindi-client'
```

Attached is an example script that just sets some simple properties with the CCD Simulator.  In my indi-allsky project, I get segfaults or recursion errors, but the code for setting properties is more complex than my example below.

Anytime a property is changed, I receive an exception like the following.

```
2022-11-25 21:32:36,224 [INFO] MainProcess __init__() #36: INDI version: 1.9.9
2022-11-25 21:32:36,224 [INFO] MainProcess <module>() #81: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2022-11-25 21:32:36,225 [INFO] MainProcess <module>() #94: Connecting to CCD Simulator
2022-11-25 21:32:36,227 [INFO] MainProcess newDevice() #40: new device Telescope Simulator
2022-11-25 21:32:36,228 [INFO] MainProcess newDevice() #40: new device Telescope Simulator
2022-11-25 21:32:36,229 [INFO] MainProcess newDevice() #40: new device CCD Simulator
2022-11-25 21:32:36,229 [INFO] MainProcess newDevice() #40: new device CCD Simulator
2022-11-25 21:32:36,726 [INFO] MainProcess <module>() #94: Connecting to CCD Simulator
2022-11-25 21:32:37,228 [INFO] MainProcess <module>() #100: Get CONNECTION control
2022-11-25 21:32:37,730 [INFO] MainProcess <module>() #114: ccd connected
2022-11-25 21:32:37,731 [INFO] MainProcess <module>() #121: Get EQUITORIAL_PE control (number)
2022-11-25 21:32:38,232 [INFO] MainProcess <module>() #126: Set EQUATORIAL_PE to M13
Traceback (most recent call last):
  File "/home/aaron/git/indi-allsky/./testing/indi_basic.py", line 130, in <module>
    indiclient.sendNewNumber(equatorial_pe)
  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/PyIndi.py", line 909, in sendNewNumber
    return _PyIndi.BaseClient_sendNewNumber(self, pp)
TypeError: in method 'BaseClient_sendNewNumber', argument 2 of type 'INDI::PropertyView< INumber > *'
```

Test script:

```
#!/usr/bin/env python3

import PyIndi
import time
import sys
import logging

CCD = "CCD Simulator"

INDI_SERVER = "localhost"
INDI_PORT = 7624

logger = logging.getLogger(__name__)
logger.setLevel(level=logging.INFO)

LOG_FORMATTER_STREAM = logging.Formatter('%(asctime)s [%(levelname)s] %(processName)s %(funcName)s() #%(lineno)d: %(message)s')
LOG_HANDLER_STREAM = logging.StreamHandler()
LOG_HANDLER_STREAM.setFormatter(LOG_FORMATTER_STREAM)
logger.addHandler(LOG_HANDLER_STREAM)

class IndiClient(PyIndi.BaseClient):
    def __init__(self):
        super(IndiClient, self).__init__()

        pyindi_version = '.'.join((
            str(getattr(PyIndi, 'INDI_VERSION_MAJOR', -1)),
            str(getattr(PyIndi, 'INDI_VERSION_MINOR', -1)),
            str(getattr(PyIndi, 'INDI_VERSION_RELEASE', -1)),
        ))

        logger.info("INDI version: %s", pyindi_version)
[... 87 lines cut ...]
### Text control test
ccd_directory_location = None
while not ccd_directory_location:
    logger.info("Get CCD_DIRECTORY_LOCATION control (text)")
    ccd_directory_location = device_ccd.getText("CCD_DIRECTORY_LOCATION")
    time.sleep(0.5)

logger.info("Set CCD_DIRECTORY_LOCATION to /tmp")

ccd_directory_location[0].text = "/tmp"  # LOCATION
indiclient.sendNewText(ccd_directory_location)

### Switch control test
simulate_bayer = None
while not simulate_bayer:
    logger.info("Get SIMULATE_BAYER control (switch)")
    simulate_bayer = device_ccd.getSwitch("SIMULATE_BAYER")
    time.sleep(0.5)

logger.info("Set SIMULATE_BAYER to Enable")

simulate_bayer[0].setState(PyIndi.ISS_ON)   # INDI_ENABLED
simulate_bayer[1].setState(PyIndi.ISS_OFF)  # INDI_DISABLED
indiclient.sendNewSwitch(simulate_bayer)
```

pawel-soja, 2022-11-26:
Fantastic! Thank you very much for the example. I'll take care of this problem.

pawel-soja, 2022-11-27:
The problem is with the implicit conversion between types, which is natural and default in C++.

I will add additional constructors to `INDI Core` to help with the conversion. In addition, a few things will need to be removed from the `indiclientpython.i` file.

@knro It will be easy for me to make changes after approval: https://github.com/indilib/indi/pull/1767

pawel-soja, 2022-11-27:
> Attached is an example script that just sets some simple properties with the CCD Simulator. In my indi-allsky project, I get segfaults or recursion errors, but the code for setting properties is more complex than my example below.

I didn't catch the segfaults.

I explain where the identified errors came from:
### RecursionError: maximum recursion depth exceeded
Inheriting from BaseClient which inherits BaseMediator, virtual methods had to be implemented.
```python
    def newDevice(self, d):
        pass

    def removeDevice(self, d):
        pass

    def newProperty(self, p):
        pass

    def removeProperty(self, p):
        pass

    def newBLOB(self, bp):
        pass

    def newSwitch(self, svp):
        pass

    def newNumber(self, nvp):
        pass

    def newText(self, tvp):
        pass

    def newLight(self, lvp):
        pass

    def newMessage(self, d, m):
        pass

    def serverConnected(self):
        pass

    def serverDisconnected(self, code):
        pass
```
The problem started when a new virtual method was added in version `1.9.9`:
```cpp
        /** @brief Emmited when a new property value arrives from INDI server.
         *  @param property Property container.
         */
        virtual void updateProperty(INDI::Property property);
```
The method was not implemented on the Python side and Swig generated a function that calls itself.
However, the new library does **not require** virtual functions to be implemented as it implements empty functions by default.

To solve the problem, and not have to implement all functions if not needed, you should also mark the BaseMediator as (similar to BaseClient):
```
%feature("director") BaseMediator;
```

### TypeError: in method 'BaseClient_sendNewNumber', argument 2 of type
I thought there was a problem with assigning a `PropertyXXX` value type to a `Property` type, which shouldn't be because `PropertyXXX` inherits from `Property`. 

The culprit of this problem is the extra lines:
https://github.com/indilib/pyindi-client/blob/ffd939bebabdb5423cd1666f84288eafcb0996ff/indiclientpython.i#L222-L231

After removing them, the problem disappears.

I guess where they came from, in version `1.9.8` (basedevice.h):
```cpp
        /** @return Return vector number property given its name */
        INDI::PropertyView<INumber> *getNumber(const char *name) const;
        /** @return Return vector text property given its name */
        INDI::PropertyView<IText>   *getText(const char *name) const;
        /** @return Return vector switch property given its name */
        INDI::PropertyView<ISwitch> *getSwitch(const char *name) const;
        /** @return Return vector light property given its name */
        INDI::PropertyView<ILight>  *getLight(const char *name) const;
        /** @return Return vector BLOB property given its name */
        INDI::PropertyView<IBLOB>   *getBLOB(const char *name) const;
```

For version `1.9.9` low-level magic types are slowly disappearing (basedevice.h):
```cpp
        /** @return Return vector number property given its name */
        INDI::PropertyNumber getNumber(const char *name) const;
        /** @return Return vector text property given its name */
        INDI::PropertyText getText(const char *name) const;
        /** @return Return vector switch property given its name */
        INDI::PropertySwitch getSwitch(const char *name) const;
        /** @return Return vector light property given its name */
        INDI::PropertyLight getLight(const char *name) const;
        /** @return Return vector BLOB property given its name */
        INDI::PropertyBlob getBLOB(const char *name) const;
```

There are still places where the previous types are behind, but I will try to use the new ones as soon as possible and add the appropriate constructors to the `1.9.9` version.

### Summary
Soon I will upload appropriate pull requests to both repositories, I am currently testing if everything is ok.

aaronwmorris, 2022-12-01:
This is a more complex method of setting properties that allows setting property values by names instead of just manually setting the indexes.  I left the unneeded methods in for testing with older versions of indi.

This is the segfault I receive:
```
$ ./indi_complex.py 
2022-11-30 21:48:11,660 [INFO] MainProcess __init__() #75: creating an instance of IndiClient
2022-11-30 21:48:11,661 [INFO] MainProcess <module>() #288: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2022-11-30 21:48:11,661 [INFO] MainProcess serverConnected() #121: Server connected (localhost:7624)
2022-11-30 21:48:11,662 [INFO] MainProcess newDevice() #79: new device CCD Simulator
2022-11-30 21:48:11,663 [INFO] MainProcess newDevice() #79: new device CCD Simulator
2022-11-30 21:48:11,664 [INFO] MainProcess newDevice() #79: new device Telescope Simulator
2022-11-30 21:48:11,665 [INFO] MainProcess newDevice() #79: new device Telescope Simulator
2022-11-30 21:48:19,670 [INFO] MainProcess findCcds() #144: Found device CCD Simulator
2022-11-30 21:48:19,671 [INFO] MainProcess findCcds() #149:  Detected ccd
2022-11-30 21:48:19,672 [INFO] MainProcess findCcds() #149:  Detected guider
2022-11-30 21:48:19,672 [INFO] MainProcess findCcds() #149:  Detected filter
2022-11-30 21:48:19,673 [INFO] MainProcess findCcds() #144: Found device Telescope Simulator
2022-11-30 21:48:19,673 [INFO] MainProcess findCcds() #149:  Detected telescope
2022-11-30 21:48:19,674 [INFO] MainProcess findCcds() #149:  Detected guider
2022-11-30 21:48:19,675 [INFO] MainProcess <module>() #306: Found 1 CCDs
2022-11-30 21:48:19,675 [WARNING] MainProcess <module>() #309: Connecting to device CCD Simulator
2022-11-30 21:48:19,675 [INFO] MainProcess configureDevice() #159: Setting switch SIMULATE_BAYER
Segmentation fault
```

```
#!/usr/bin/env python3

import sys
import logging
import time
from collections import OrderedDict
import ctypes
import PyIndi

INDI_SERVER = 'localhost'
INDI_PORT = 7624

INDI_CONFIG = OrderedDict({
    "SWITCHES" : {
        "SIMULATE_BAYER" : {
            "on"  : ["INDI_ENABLED"],
            "off" : ["INDI_DISABLED"],
        },
    },
    "PROPERTIES" : {
        "EQUATORIAL_PE" : {
            "RA_PE"  : 16.7175,
            "DEC_PE" : 36.4233
        },
    },
    "TEXT" : {
        "CCD_DIRECTORY_LOCATION" : {
            "LOCATION" : "/tmp",
        },
    },
})

logger = logging.getLogger(__name__)
logger.setLevel(level=logging.INFO)

LOG_FORMATTER_STREAM = logging.Formatter('%(asctime)s [%(levelname)s] %(processName)s %(funcName)s() #%(lineno)d: %(message)s')
LOG_HANDLER_STREAM = logging.StreamHandler()
LOG_HANDLER_STREAM.setFormatter(LOG_FORMATTER_STREAM)
logger.addHandler(LOG_HANDLER_STREAM)

class IndiClient(PyIndi.BaseClient):

    __state_to_str = {
[... 224 lines cut ...]

logger.info("Connecting to indiserver")
if not indiclient.connectServer():
    logger.error("No indiserver running on %s:%d - Try to run", indiclient.getHost(), indiclient.getPort())
    logger.error("  indiserver indi_simulator_telescope indi_simulator_ccd")
    sys.exit(1)

# give devices a chance to register
time.sleep(8)

ccd_list = indiclient.findCcds()

if len(ccd_list) == 0:
    logger.error('No CCDs detected')
    time.sleep(1)
    sys.exit(1)

logger.info('Found %d CCDs', len(ccd_list))
ccdDevice = ccd_list[0]

logger.warning('Connecting to device %s', ccdDevice.getDeviceName())
indiclient.connectDevice(ccdDevice.getDeviceName())

indiclient.configureDevice(ccdDevice, INDI_CONFIG)
```

pawel-soja, 2022-12-01:
Okay I have it.
```python
            off_switches = [s.name for s in c if s.name not in on_switches]
```
When items with a loop after any object, only the `__getitem__` method is used. The ` __len__` function is not performed. In order for the loop to end at the right time, the ` __getitem__` function must quit the exception.

I added the appropriate fixes to https://github.com/indilib/pyindi-client/pull/27

aaronwmorris, 2022-12-01:
That is awesome!  You are killing it.  :-)  I tested the changes and it appears to be working.

Final question:  **With these changes to pyindi-client, will it be compatible with older versions of indi or will it require 1.9.9?**

pawel-soja, 2022-12-01:
It should be backwards compatible, although I haven't tested all versions.

I recommend using getters and setters, additionally note that there are built-in functions that return string as state or rule.

### Property Example
```python
#!/usr/bin/env python3
import PyIndi

prop = PyIndi.PropertySwitch(2)

# setters
prop.setDeviceName("Property Device Name")
prop.setName("Property Name")
prop.setLabel("Property Label")
prop.setGroupName("Property Group Name")
prop.setPermission(PyIndi.IP_RW)
prop.setTimeout(123)
prop.setState(PyIndi.IPS_BUSY)
prop.setTimestamp("1234")
prop.setRule(PyIndi.ISR_ATMOST1)

prop[0].setName("Widget Name 1")
prop[0].setLabel("Widget Label 1")
prop[0].setState(PyIndi.ISS_OFF)

prop[1].setName("Widget Name 2")
prop[1].setLabel("Widget Label 2")
prop[1].setState(PyIndi.ISS_ON)

# getters
print('---- property ----')
print('        getDeviceName:', prop.getDeviceName())
print('              getName:', prop.getName())
print('             getLabel:', prop.getLabel())
print('         getGroupName:', prop.getGroupName())
print('        getPermission:', prop.getPermission())
print('getPermissionAsString:', prop.getPermissionAsString())
print('           getTimeout:', prop.getTimeout())
print('         getTimestamp:', prop.getTimestamp())
print('              getRule:', prop.getRule())
print('      getRuleAsString:', prop.getRuleAsString())

for widget in prop:
    print('')
    print('---- widget ----')
    print('         getName:', widget.getName())
    print('        getLabel:', widget.getLabel())
    print('        getState:', widget.getState())
    print('getStateAsString:', widget.getStateAsString())

```

#### Output:
```cpp
---- property ----
        getDeviceName: Property Device Name
              getName: Property Name
             getLabel: Property Label
         getGroupName: Property Group Name
        getPermission: 2
getPermissionAsString: rw
           getTimeout: 123.0
         getTimestamp: 1234
              getRule: 1
      getRuleAsString: AtMostOne

---- widget ----
         getName: Widget Name 1
        getLabel: Widget Label 1
        getState: 0
getStateAsString: Off

---- widget ----
         getName: Widget Name 2
        getLabel: Widget Label 2
        getState: 1
getStateAsString: On
```

### Property Cast Example
```python
#!/usr/bin/env python3
import PyIndi

prop = PyIndi.Property(PyIndi.PropertySwitch(2))

print('Type:', prop.getTypeAsString())

print('Is light? ', PyIndi.PropertyLight(prop).isValid())
print('Is Switch?', PyIndi.PropertySwitch(prop).isValid())
print('Is Number?', PyIndi.PropertyNumber(prop).isValid())
```

#### Output
```
Type: INDI_SWITCH
Is light?  False
Is Switch? True
Is Number? False
```

Conversion can be useful when reading a given property through new methods in the mediator (now 3 methods are enough to handle all cases for a property).
https://github.com/indilib/indi/blob/8d9ba364579804b0eebc5f3971844a34ac056d32/libs/indidevice/indibase.h#L105-L119
```cpp
    public:
        /** @brief Emmited when a new property is created for an INDI driver.
         *  @param property Property container.
         */
        virtual void newProperty(INDI::Property property);

        /** @brief Emmited when a new property value arrives from INDI server.
         *  @param property Property container.
         */
        virtual void updateProperty(INDI::Property property);

        /** @brief Emmited when a property is deleted for an INDI driver.
         *  @param property Property container.
         */
        virtual void removeProperty(INDI::Property property);
```

aaronwmorris, 2022-12-02:
Very interesting information.

I had already converted some of the code to use getters and setters since direct access to some of the object properties no longer worked.  Looks like I need to finish that work.

aaronwmorris, 2022-12-06:
Looks like this fixes indi 1.9.9, unfortunately it is not backwards compatible with indi 1.9.8 or 1.9.7.

I will submit a PR to increment the version number of pyindi-client.

closed 2022-12-06 by aaronwmorris