indilib/pyindi-client #30: Indi client test TypeError: 'SwigPyObject' object is not iterable

f6bvp, 2023-01-01:
Running a sample Pyindi client test I get TypeError: 'SwigPyObject' object is not iterable

Here is the full report :

Connecting and waiting 1 sec
INDI::BaseClient::connectServer: creating new connection...
2023-01-01 23:11:19,264 Server connected (www1.f4klo.ampr.org:7624)
INDI::BaseClient::connectServer: Already connected.
2023-01-01 23:11:19,266 new device Radiotelescope La Villette
2023-01-01 23:11:19,266 new property CONNECTION for device Radiotelescope La Villette
2023-01-01 23:11:19,267 new property DRIVER_INFO for device Radiotelescope La Villette
2023-01-01 23:11:19,269 new property POLLING_PERIOD for device Radiotelescope La Villette
2023-01-01 23:11:19,269 new property DEBUG for device Radiotelescope La Villette
2023-01-01 23:11:19,269 new property CONFIG_PROCESS for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property CONNECTION_MODE for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property SYSTEM_PORTS for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property DEVICE_PORT for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property DEVICE_BAUD_RATE for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property DEVICE_AUTO_SEARCH for device Radiotelescope La Villette
2023-01-01 23:11:19,270 new property DEVICE_PORT_SCAN for device Radiotelescope La Villette
2023-01-01 23:11:19,271 new property ACTIVE_DEVICES for device Radiotelescope La Villette
2023-01-01 23:11:19,271 new property DOME_POLICY for device Radiotelescope La Villette
2023-01-01 23:11:19,271 new property TELESCOPE_INFO for device Radiotelescope La Villette
2023-01-01 23:11:19,271 new property SCOPE_CONFIG_NAME for device Radiotelescope La Villette
2023-01-01 23:11:19,272 new Text ACTIVE_DEVICES for device Radiotelescope La Villette
2023-01-01 23:11:19,276 new Number TELESCOPE_INFO for device Radiotelescope La Villette
2023-01-01 23:11:19,276 new Text SCOPE_CONFIG_NAME for device Radiotelescope La Villette
List of devices
Radiotelescope La Villette
List of Device Properties
-- Radiotelescope La Villette
Traceback (most recent call last):
  File "/home/f4klo/Pyindi/planet/./Pyindi_client_test.py", line 137, in <module>
    for p in lp:
TypeError: 'SwigPyObject' object is not iterable

Bernard

f6bvp, 2023-01-01:
There is another error that may not be related directly to Pyindi 

Traceback (most recent call last):
  File "/home/f4klo/Pyindi/planet/./Pyindi_client_simulator_test.py", line 137, in <module>
    for p in lp:
TypeError: 'SwigPyObject' object is not iterable
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.

knro, 2025-05-04:
We released an update version, please check.

f6bvp, 2025-05-04:
Bonjour,Deux ans après mon rapport de bug je reçois ce correctif.Jasem Mutlaq, l’auteur principal de INDI n’explique pas en quoi consiste la mise à jour. Est-ce toute la librairie ?Je vais regarder si nos clients Pyindi sont réellement concernés ou simplement le test.73 de Bernard, F6BVP

Dr Bernard Pidoux73 de Bernard, F6BVP

closed 2025-05-04 by knro