indilib/pyindi-client #31: Error while connecting the INDI device

AstroAir, 2023-01-04:
First,I'm sorry to bother you.But I've been confused recently.I'm trying to write a INDI Python client and use pyindi.However,when I tried to connect a device.There would always be this error.

`Error in sys.excepthook:
 Original exception was:`

I can scan the device and get the name of the device, but every time I connect, an error occurs.
I also found that the provided examples are almost unusable, and many strange errors will occur.

This function `getDriverInterface()` will return : `ITEXT No DRIVERINTERFACE`

I hope to have an available example.Sorry again for disturbing you.But I really want to get the answer.

AstroAir, 2023-01-04:
In addition, I also tried to communicate with the INDI server directly through the socket. I sent `<getProperties version='1.7'/>` to server.And I also received the results normally, but I don't know when to stop accepting. There is no exact ending sign like `</root>`.

knro, 2023-01-04:
@pawel-soja IIRC, this function removed recently or moved?

AstroAir, 2023-01-04:
Now I found some new problems.When I first use the example in README.md , everything worked well.However , after I restart the INDI server . Some error occured.
`Error in sys.excepthook:
Original exception was:
Exception ignored in m_clear of module _signal`

pawel-soja, 2023-01-04:
@knro functionality should be as it was. Maybe I'm missing something, please clarify your question.

@AstroAir I cannot reproduce the problem.
Please provide additional information:
- system distribution and version
- Indi library version or commit hash
- PyIndi version or commit hash
- python version
- sample python code

AstroAir, 2023-01-04:
@pawel-soja 
INDI lib version 1.9,9,
Pyindi version 1.9.1 (I built by myself,not using pip)
System : Ubuntu 22.04 wsl1
Python version : 3.10.6

indiclient.py

`
import ctypes
import time
from utils.lightlog import lightlog
log = lightlog(__name__)
from utils.i18n import _

import PyIndi

class INDIDevice():
    """
        Basic INDI Device Interface
    """

    timeout = 30

    def __init__(self , device_name : str , indi_client) -> None:
        """
            Initialize the INDI Device | 初始化INID设备
            Args : 
                device_name : str # name of the device
                indi_client : INDIClient
            Returns : None
        """
        self.device_name = device_name
        self.indi_client = indi_client
        self.device = None

    def connect(self) -> dict:
        """
            Connect to the INDI Device | 连接INDI设备
            Args : None
            Returns : dict
        """
        if self.is_connected:
            log.logw(_("Device had already connected , please do not connect again"))
            return log.return_error(_("Device has already connected"),{})

    def scanning(self) -> None:
        """
            Scan the INDI device | 扫描INDI设备
            Args : None
            Returns : None
        """
        # If the device is not found, there is looping
        while not self.device:
            self.device = self.indi_client.getDevice(self.name)

    def get_control(self,name : str , control_type : str , timeout = timeout) -> dict | None:
        """
            Get the value of the control | 返回值
            Args : 
                name : str # name of the control
                control_type : str # type of the control
                timeout : int # timeout
            Returns : 
                dict | None
        """
        control = None
        attr = {
            'number': 'getNumber',
            'switch': 'getSwitch',
[... 262 lines cut ...]
        else:
            interface.acquire()
            device_interfaces = int(ctypes.cast(interface.__int__(), ctypes.POINTER(ctypes.c_uint16)).contents.value)
            interface.disown()

        interfaces = {
            PyIndi.BaseDevice.GENERAL_INTERFACE: 'general', 
            PyIndi.BaseDevice.TELESCOPE_INTERFACE: 'telescope',
            PyIndi.BaseDevice.CCD_INTERFACE: 'ccd',
            PyIndi.BaseDevice.GUIDER_INTERFACE: 'guider',
            PyIndi.BaseDevice.FOCUSER_INTERFACE: 'focuser',
            PyIndi.BaseDevice.FILTER_INTERFACE: 'filter',
            PyIndi.BaseDevice.DOME_INTERFACE: 'dome',
            PyIndi.BaseDevice.GPS_INTERFACE: 'gps',
            PyIndi.BaseDevice.WEATHER_INTERFACE: 'weather',
            PyIndi.BaseDevice.AO_INTERFACE: 'ao',
            PyIndi.BaseDevice.DUSTCAP_INTERFACE: 'dustcap',
            PyIndi.BaseDevice.LIGHTBOX_INTERFACE: 'lightbox',
            PyIndi.BaseDevice.DETECTOR_INTERFACE: 'detector',
            PyIndi.BaseDevice.ROTATOR_INTERFACE: 'rotator',
            PyIndi.BaseDevice.AUX_INTERFACE: 'aux'
        }
        interfaces = [interfaces[x] for x in interfaces if x & device_interfaces]
        return interfaces
`

closed 2023-01-07 by AstroAir