indilib/pyindi-client #32: INDI 2.0.0 - newBLOB() handler not firing

aaronwmorris, 2023-02-01:
Under indi 2.0.0 it appears the newBLOB() handler is no longer being called when a blob is returned by a camera.

Is this a bug or is there some new method for accessing the blobs returned by the camera?

knro, 2023-02-02:
With INDI 2.0.0, @pawel-soja added updateProperty function now, which may be used to access an updated BLOB. This requires some changes on pyindi-client side to support?

pawel-soja, 2023-02-02:
From version 2.0.0, the methods to be implemented are more generic and unified.

### Source interface in C++
```cpp
/** @class INDI::BaseMediator
 *  @brief Meditates event notification as generated by driver and passed to clients.
 */
class INDI::BaseMediator
{
    public:
        virtual ~BaseMediator() = default;

    public:
        /** @brief Emmited when a new device is created from INDI server.
         *  @param baseDevice BaseDevice instance.
         */
        virtual void newDevice(INDI::BaseDevice baseDevice);

        /** @brief Emmited when a device is deleted from INDI server.
         *  @param baseDevice BaseDevice instance.
         */
        virtual void removeDevice(INDI::BaseDevice baseDevice);        

    public:
        /** @brief Emmited when a new property is created for an INDI driver.
         *  @param property Property container.
         */
        virtual void newProperty(INDI::Property property);

        /** @brief Emmited when a new property value arrives from INDI server.
         *  @param property Property container.
         */
        virtual void updateProperty(INDI::Property property);

        /** @brief Emmited when a property is deleted for an INDI driver.
         *  @param property Property container.
         */
        virtual void removeProperty(INDI::Property property);

    public:
        /** @brief Emmited when a new message arrives from INDI server.
         *  @param baseDevice BaseDevice instance the message is sent to.
         *  @param messageID ID of the message that can be used to retrieve the message from the device's messageQueue() function.
         */
        virtual void newMessage(INDI::BaseDevice baseDevice, int messageID);

    public:
        /** @brief Emmited when the server is connected. */
        virtual void serverConnected();

        /** @brief Emmited when the server gets disconnected.
         *  @param exit_code 0 if client was requested to disconnect from server. -1 if connection to server is terminated due to remote server disconnection.
         */
        virtual void serverDisconnected(int exit_code);

    public: // deprecated interface
#if INDI_VERSION_MAJOR < 2
        /** @brief Emmited when a new device is created from INDI server.
         *  @param dp Pointer to the base device instance
         */
        virtual void newDevice(INDI::BaseDevice *dp); // deprecated

        /** @brief Emmited when a device is deleted from INDI server.
         *  @param dp Pointer to the base device instance.
         */
        virtual void removeDevice(INDI::BaseDevice *dp); // deprecated

        /** @brief Emmited when a new property is created for an INDI driver.
         *  @param property Pointer to the Property Container
         */
[... 153 lines cut ...]
INDI::BaseClient::connectServer: creating new connection...
New device: CCD Simulator
New property: CONNECTION
New property: DRIVER_INFO
New property: POLLING_PERIOD
New property: DEBUG
New property: CONFIG_PROCESS
New property: DEBUG_LEVEL
New property: LOGGING_LEVEL
...
New property: FILTER_SLOT
New property: FILTER_NAME
Update property: CCD_EXPOSURE
Update property: CCD_EXPOSURE
Update property: CCD_EXPOSURE
Update property: CCD1
      name: CCD1
     label: Image Data
    device: CCD Simulator
    format: .fits
  blob len: 2626560
      size: 2626560
Update property: CCD_EXPOSURE

```

knro, 2023-02-02:
@pawel-soja Maybe good idea to add above to some 2.0.0 migration guide and make it available in the repo top level with a note in README? What do you think?

knro, 2023-02-02:
I don't think we need to go into too much detail, just what you wrote above is now enough. I can even add it myself if you don't have time. The only affected client I know about now is Stellarium, and I just forked it so I can perform the fixes myself and send them a PR.

aaronwmorris, 2023-02-02:
I found that this seems to work as a simple method to maintain compatibility between indi 1.x.x and 2.x.x

```
    # Emmited when a new property value arrives from INDI server.
    def updateProperty(self, prop):
        self.logger.info("Update property: %s", prop.getName())

        if prop.isNameMatch("CCD1"):
            blob = PyIndi.PropertyBlob(prop) # cast generic type to blob type

            # usually, the first item is the expected photo
            self.newBLOB(blob[0])

    def newBLOB(self, bp):
        self.logger.info("new BLOB %s", bp.name)
```

pawel-soja, 2023-02-02:
@aaronwmorris Thanks, this is also the solution I proposed for PHD2.
https://github.com/indilib/indi/issues/1833

And this is a good solution for python.
Here it's so much easier that such a solution is acceptable in the long run, because there are no types here and raw indicators disappear.

```python
class IndiClient(PyIndi.BaseClient):
    def __init__(self):
        super(IndiClient, self).__init__()

    # Call functions in old style 
    def updateProperty(self, prop):
        if prop.getType() == PyIndi.INDI_NUMBER:
            self.newNumber(PyIndi.PropertyNumber(prop))
        elif prop.getType() == PyIndi.INDI_SWITCH:
            self.newSwitch(PyIndi.PropertySwitch(prop))
        elif prop.getType() == PyIndi.INDI_TEXT:
            self.newText(PyIndi.PropertyText(prop))
        elif prop.getType() == PyIndi.INDI_LIGHT:
            self.newLight(PyIndi.PropertyLight(prop))
        elif prop.getType() == PyIndi.INDI_BLOB:
            self.newBLOB(PyIndi.PropertyBlob(prop)[0])

    # The methods work again with INDI Core 2.0.0!
    def newBLOB(self, prop):
        print(f"new BLOB {prop.getName()}")
    def newSwitch(self, prop):
        print(f"new Switch {prop.getName()} for device {prop.getDeviceName()}")
    def newNumber(self, prop):
        print(f"new Number {prop.getName()} for device {prop.getDeviceName()}")
    def newText(self, prop):
        print(f"new Text {prop.getName()} for device {prop.getDeviceName()}")
    def newLight(self, prop):
        print(f"new Light {prop.getName()} for device {prop.getDeviceName()}")
```

pawel-soja, 2023-02-03:
Pull request https://github.com/indilib/pyindi-client/pull/33

Guys, can you take care of the next steps related to the pull request? I didn't change the version, etc. I see that something else is hanging in pull requests and I'm not sure how you want to arrange it.
Improved examples, tests and readme. If there's anything else I can add or clarify, please let me know.

aaronwmorris, 2023-02-03:
I have discovered that if you are running 1.9.9 both the updateProperty() and newBLOB() methods will be called.  If both are implemented, blobs will be processed twice if they both use the same code path.

There is nothing that can really be done about this from the indi side since the code is already in the wild, so this condition would have to be managed in the client.

aaronwmorris, 2023-02-03:
That would work, although I might go with something a little different style-wise.  Either way, it works.

```python
    # Emmited when a new property value arrives from INDI server.
    def updateProperty(self, prop):
        if not hasattr(PyIndi.BaseMediator, 'newNumber'):
            # indi 1.x bypass
            return

        if prop.getType() == PyIndi.INDI_NUMBER:
            self.newNumber(PyIndi.PropertyNumber(prop))
        elif prop.getType() == PyIndi.INDI_SWITCH:
            self.newSwitch(PyIndi.PropertySwitch(prop))
        elif prop.getType() == PyIndi.INDI_TEXT:
            self.newText(PyIndi.PropertyText(prop))
        elif prop.getType() == PyIndi.INDI_LIGHT:
            self.newLight(PyIndi.PropertyLight(prop))
        elif prop.getType() == PyIndi.INDI_BLOB:
            self.newBLOB(PyIndi.PropertyBlob(prop)[0])
```

closed 2024-02-02 by aaronwmorris