indilib/pyindi-client #37: Impossible to build/install pyindi-client on my system

WLR86, 2023-06-23:
Trying to build/install pyindi-client on a Raspberry Pi Zero W running Raspbian 11.7, having these packages installed : 
```
ii  indi-bin                             1.8.8+dfsg-1
ii  libindi-data                         1.8.8+dfsg-1
ii  libindi-dev                          1.8.8+dfsg-1
ii  libindi-plugins:armhf                1.8.8+dfsg-1
ii  libindialignmentdriver1:armhf        1.8.8+dfsg-1
ii  libindidriver1:armhf                 1.8.8+dfsg-1
ii  libindilx200-1:armhf                 1.8.8+dfsg-1
```
Output of installation using pip3 : 
```
$ pip3 install pyindi-client
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Using cached pyindi-client-1.9.1.tar.gz (18 kB)
Building wheels for collected packages: pyindi-client
  Building wheel for pyindi-client (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-aaaci_4f
       cwd: /tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/
  Complete output (30 lines):
  /usr/lib/python3/dist-packages/setuptools/dist.py:461: UserWarning: Normalizing 'v1.9.1' to '1.9.1'
    warnings.warn(tmpl.format(**locals()))
  running bdist_wheel
  running build
  running build_ext
  building '_PyIndi' extension
  swigging indiclientpython.i to indiclientpython_wrap.cpp
  swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
  indiclientpython.i:41: Error: Unable to find 'indimacros.h'
  indiclientpython.i:47: Error: Unable to find 'indiwidgettraits.h'
  indiclientpython.i:56: Error: Unable to find 'indipropertyview.h'
  indiclientpython.i:104: Error: Unable to find 'indipropertybasic.h'
  indiclientpython.i:122: Error: Unable to find 'indipropertytext.h'
  indiclientpython.i:123: Error: Unable to find 'indipropertynumber.h'
  indiclientpython.i:124: Error: Unable to find 'indipropertyswitch.h'
  indiclientpython.i:125: Error: Unable to find 'indipropertylight.h'
  indiclientpython.i:126: Error: Unable to find 'indipropertyblob.h'
  indiclientpython.i:196: Error: Unable to find 'indiproperties.h'
  Language subdirectory: python
  Search paths:
     ./
     /usr/include/
     /usr/include/libindi/
     /usr/local/include/libindi/
     ./swig_lib/python/
     /usr/share/swig4.0/python/
     ./swig_lib/
     /usr/share/swig4.0/
  Preprocessing...
  error: command '/usr/bin/swig' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for pyindi-client
  Running setup.py clean for pyindi-client
Failed to build pyindi-client
Installing collected packages: pyindi-client
    Running setup.py install for pyindi-client ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-dwsuavba/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/willy/.local/include/python3.9/pyindi-client
         cwd: /tmp/pip-install-r1r0on5j/pyindi-client_6e15eda540894963b9638d84d90320bf/
    Complete output (29 lines):
    /usr/lib/python3/dist-packages/setuptools/dist.py:461: UserWarning: Normalizing 'v1.9.1' to '1.9.1'
      warnings.warn(tmpl.format(**locals()))
    running install
    running build_ext
    building '_PyIndi' extension
    swigging indiclientpython.i to indiclientpython_wrap.cpp
    swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
    indiclientpython.i:41: Error: Unable to find 'indimacros.h'
[... 81 lines cut ...]
indiclientpython_wrap.cpp:62642:78: error: expected ')' before ';' token
62642 |     return (void *)((INDI::BaseMediator *)  ((INDI::AbstractBaseClient *) x));
      |                    ~                                                         ^
      |                                                                              )
indiclientpython_wrap.cpp: In function 'void* _p_INDI__BaseClientTo_p_INDI__BaseMediator(void*, int*)':
indiclientpython_wrap.cpp:62645:51: error: 'AbstractBaseClient' is not a member of 'INDI'
62645 |     return (void *)((INDI::BaseMediator *) (INDI::AbstractBaseClient *) ((INDI::BaseClient *) x));
      |                                                   ^~~~~~~~~~~~~~~~~~
indiclientpython_wrap.cpp:62645:71: error: expected primary-expression before ')' token
62645 |     return (void *)((INDI::BaseMediator *) (INDI::AbstractBaseClient *) ((INDI::BaseClient *) x));
      |                                                                       ^
indiclientpython_wrap.cpp: In function 'PyObject* PyInit__PyIndi()':
indiclientpython_wrap.cpp:982:65: warning: 'void PyEval_InitThreads()' is deprecated [-Wdeprecated-declarations]
  982 | #     define SWIG_PYTHON_INITIALIZE_THREADS  PyEval_InitThreads()
      |                                                                 ^
indiclientpython_wrap.cpp:63884:3: note: in expansion of macro 'SWIG_PYTHON_INITIALIZE_THREADS'
63884 |   SWIG_PYTHON_INITIALIZE_THREADS;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from indiclientpython_wrap.cpp:180:
/usr/include/python3.9/ceval.h:130:37: note: declared here
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~
error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
```

KraisinS, 2025-05-02:
Has the same problem this week, trying to install pyindi-client. The header cannot be found, even try to search the entire machine.  Any way to solve this?  Thank you,

knro, 2025-05-04:
This might be related to using Bullseye which comes with older python version. Try to use pip with --no-cache to force download of the latest released pyindi-client package to see if this helps.