indilib/pyindi-client #38: Custom path for libindiclient.a

jodier, 2023-07-13:
Dear pyindi-client deceloppers,

Would it be possible to implement something to make it possible to specify a custom libindiclient.a path when installing pyindi-client?

It's a very interesting feature to embed "libindiclient.a" and "pyindi-client" in an application and not only in "/usr/lib/" + march, "/usr/lib", "/usr/lib64", "/lib", "/lib64", "/usr/local/lib/" + march and "/usr/local/lib".

To do so (and be compatible with both a "python setup.py install" and "pip install pyindi-client"), the solution is to add an environment variable (for example INDICLIENT_PATH):

...

if "INDICLIENT_PATH" in os.environs:
    libindisearchpaths.append(os.environs["INDICLIENT_PATH"])

libindipath = ""

...

Best regards,
Jérôme

tbowers7, 2024-06-04:
I second a request for this feature.

knro, 2024-06-05:
Can anyone submit a PR for this? It would be greatly appreciated.