indilib/pyindi-client #41: 'BaseClient' object has no attribute 'newText'  and  RecursionError: maximum recursion depth exceeded

Akinetopsia1, 2023-08-26:
None of the examples work on Raspberry Pi with Raspbian GNU/Linux 10 (buster), Kernel: Linux 5.10.103-v7l+

**testindiclient.py** starts listing props an then fails with: 

`2023-08-25 23:46:08,453 new property TELESCOPE_TIMED_GUIDE_NS as INDI_NUMBER for device Telescope Simulator
2023-08-25 23:46:08,454 new property TELESCOPE_TIMED_GUIDE_WE as INDI_NUMBER for device Telescope Simulator
2023-08-25 23:46:08,454 new property GUIDE_RATE as INDI_NUMBER for device Telescope Simulator
Traceback (most recent call last):
  File "/home/astroberry/.local/lib/python3.7/site-packages/PyIndi.py", line 1093, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BaseClient, name)
  File "/home/astroberry/.local/lib/python3.7/site-packages/PyIndi.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BaseClient' object has no attribute 'newText'`

vega.py and even tuto-minimal.py gives :

`INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
RecursionError: maximum recursion depth exceeded`

tuto-monitor.py gives:

`INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
New device  Telescope Simulator
Traceback (most recent call last):
  File "indy-monitor.py", line 20, in newProperty
    cmonitor = PyIndi.PropertySwitch(p)
  File "/home/astroberry/.local/lib/python3.7/site-packages/PyIndi.py", line 3129, in __init__
RecursionError: maximum recursion depth exceeded`

knro, 2023-08-26:
Most likely because you don't have INDI v2.0.0+ on there?

Akinetopsia1, 2023-08-26:
hmm, i would've expected an unmet requirements check and warning while installing.

I'm using the astroberry distro and it's a bit behind. INDI 1.9.7 is the latest version for it. Upgrading is a bit of a can of worms as last time i tried i had to recompile K-Stars/Ekos, had a bunch of instabilities and i would've had to also compile QT5 IIRC which is a really bad idea on a pi to say the least. 

I'm going to leave it at that and check if i can find a previous commit that works, otherwise i guess i'll work around using indi_getprop / indi_setprop

Thx for the tip

closed 2023-11-10 by Akinetopsia1