indilib/pyindi-client #43: Cannot install pyindi-client on Ubuntu 22.04

TD-SMS, 2023-10-10:
Dear Indi community and PyIndi developpers,

I am a new user of Indi, and I met some problems while trying to install PyIndi on my Ubuntu 22.04.

Here is what I've done :
- sudo apt-add-repository ppa:mutlaqja/ppa 
- sudo apt-get install indi-full gsc kstars-bleeding

The installed versions are :
- indi-full : 2.0.4+202310021623~ubuntu22.04.1
- gsc : 1.3+202212170706~ubuntu22.04.1
- kstars-bleeding : 6:3.6.7+202310090602~ubuntu22.04.1

Then :
- sudo apt-get install python-setuptools python3-dev libindi-dev swig  libcfitsio-dev  libnova-dev

libindi-dev version is :  2.0.4+202309301936~ubuntu22.04.1

Finally :
- pip3 install pyindi-client

Unfortunately, this ends with the following message : 
_/usr/include/libindi/indibase.h:10: Error: Syntax error in input(1).
      error: command '/usr/bin/swig' failed with exit code 1
      [end of output]_

The manual method (downloading the repo and "python setup.py install") give the same error (I tired with release 1.9.1 and also with current main branch)

Is there something I am doing wrong or forgetting, a compatibility problem, or is it a real bug ?

Thank you in advance for your answers and suggestions.

Best regards

knro, 2023-10-10:
This is a known issue. For now you can quickly edit /usr/include/libindi/indibase.h and change

`% include "indiapi.h"`

to 

`%include "indiapi.h"`

So just remove the space and save the file.

TD-SMS, 2023-10-11:
Dear Jasem,

Thank you very much for your help.
After the modification in indibase.h, a "sudo python setup.py install" ended successfully.

Best regards

TD-SMS, 2023-10-11:
Closing : suggested solution worked

closed 2023-10-11 by TD-SMS

KMilhan, 2024-01-09:
@knro, Looks like `indilibase.h` has been updated but the problem still exist. I basically followed what OP did.

```text
❯ pip install pyindi-client
Collecting pyindi-client
  Downloading pyindi-client-1.9.1.tar.gz (18 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pyindi-client
  Building wheel for pyindi-client (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyindi-client (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [908 lines of output]
      /tmp/pip-build-env-3lsp8go1/overlay/lib/python3.12/site-packages/setuptools/dist.py:314: InformationOnly: Normalizing 'v1.9.1' to '1.9.1'
        self.metadata.version = self._normalize_version(self.metadata.version)
      running bdist_wheel
      running build
      running build_ext
      building '_PyIndi' extension
      swigging indiclientpython.i to indiclientpython_wrap.cpp
      swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
      Language subdirectory: python
      Search paths:
         ./
         /usr/include/
         /usr/include/libindi/
         /usr/local/include/libindi/
         ./swig_lib/python/
         /usr/share/swig4.0/python/
         ./swig_lib/
         /usr/share/swig4.0/
      Preprocessing...
      Starting language-specific parse...
      /usr/include/libindi/indiwidgettraits.h:38: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:49: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:60: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:69: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:83: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indipropertyview.h:267: Warning 302: Identifier 'fill' redefined (ignored),
      /usr/include/libindi/indipropertyview.h:261: Warning 302: previous definition of 'fill'.
      /usr/include/libindi/indipropertyview.h:285: Warning 302: Identifier 'fill' redefined (ignored),
      /usr/include/libindi/indipropertyview.h:261: Warning 302: previous definition of 'fill'.
      /usr/include/libindi/indipropertyview.h:373: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:378: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:510: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:515: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:674: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:679: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:803: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:808: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:932: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:937: Warning 362: operator= ignored
      /usr/include/libindi/indiproperty.h:242: Warning 314: 'self' is a python keyword, renaming to '_self'
      /usr/include/libindi/indipropertybasic.h:131: Warning 389: operator[] ignored (consider using %extend)
      /usr/include/libindi/indipropertyswitch.h:39: Warning 325: Nested struct not currently supported (NewValues ignored)
      indiclientpython.i:181: Warning 326: Deprecated %extend name used - the struct name '_IBLOB' should be used instead of the typedef name 'IBLOB'.
      /usr/include/libindi/indiproperties.h:61: Warning 389: operator[] ignored (consider using %extend)
      /usr/include/libindi/indiproperties.h:62: Warning 389: operator[] ignored (consider using %extend)
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Nothing known about base class 'WidgetTraits< _IText >::PropertyType'. Ignored.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Maybe you forgot to instantiate 'WidgetTraits< _IText >::PropertyType' using %template.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Nothing known about base class 'WidgetTraits< _INumber >::PropertyType'. Ignored.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Maybe you forgot to instantiate 'WidgetTraits< _INumber >::PropertyType' using %template.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Nothing known about base class 'WidgetTraits< _ISwitch >::PropertyType'. Ignored.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Maybe you forgot to instantiate 'WidgetTraits< _ISwitch >::PropertyType' using %template.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Nothing known about base class 'WidgetTraits< _ILight >::PropertyType'. Ignored.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Maybe you forgot to instantiate 'WidgetTraits< _ILight >::PropertyType' using %template.
      /usr/include/libindi/indipropertyview.h:1: Warning 401: Nothing known about base class 'WidgetTraits< _IBLOB >::PropertyType'. Ignored.
[... 840 lines cut ...]
[... cut ...]
      Generating wrappers...
      error: command '/usr/bin/swig' failed with exit code 70
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyindi-client
Failed to build pyindi-client
ERROR: Could not build wheels for pyindi-client, which is required to install pyproject.toml-based projects

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: pip install --upgrade pip

```

KMilhan, 2024-01-09:
@TD-SMS , could you please test again and reopen this issue if it occurs to you again?

christianwbrock, 2024-03-02:
Hi @KMilhan,

I got the same errors as @TD-SMS 

I'm using "Ubuntu 22.04.4 LTS" with "Python 3.10.12".
My indilib comes from  https://ppa.launchpadcontent.net/mutlaqja/ppa/ubuntu.

-_Chr.