indilib/pyindi-client #44: error: use of class template 'INDI::WidgetView' requires template arguments

lukashermann, 2024-03-26:
This error happens when I try to install pyindi-client on MacOS 14.
I attached the full log output here:
[full_error.txt](https://github.com/indilib/pyindi-client/files/14761669/full_error.txt)

```
indiclientpython_wrap.cpp:37719:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37733:34: error: use of class template 'INDI::WidgetView' requires template arguments
  arg1 = reinterpret_cast< INDI::WidgetView * >(argp1);
                                 ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37738:84: error: use of class template 'INDI::WidgetView' requires template arguments
        result = (INDI::WidgetView< IText > *)new INDI::WidgetView< IText >((INDI::WidgetView const &)*arg1);
                                                                                   ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37754:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37757:26: error: use of class template 'INDI::WidgetView' requires template arguments
  std::unique_ptr< INDI::WidgetView > rvrdeleter1 ;
                         ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37773:34: error: use of class template 'INDI::WidgetView' requires template arguments
  arg1 = reinterpret_cast< INDI::WidgetView * >(argp1);
                                 ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:37779:84: error: use of class template 'INDI::WidgetView' requires template arguments
        result = (INDI::WidgetView< IText > *)new INDI::WidgetView< IText >((INDI::WidgetView &&)*arg1);
                                                                                   ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39201:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39215:34: error: use of class template 'INDI::WidgetView' requires template arguments
  arg1 = reinterpret_cast< INDI::WidgetView * >(argp1);
                                 ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39220:88: error: use of class template 'INDI::WidgetView' requires template arguments
        result = (INDI::WidgetView< INumber > *)new INDI::WidgetView< INumber >((INDI::WidgetView const &)*arg1);
                                                                                       ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39236:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39239:26: error: use of class template 'INDI::WidgetView' requires template arguments
  std::unique_ptr< INDI::WidgetView > rvrdeleter1 ;
                         ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39255:34: error: use of class template 'INDI::WidgetView' requires template arguments
  arg1 = reinterpret_cast< INDI::WidgetView * >(argp1);
                                 ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:39261:88: error: use of class template 'INDI::WidgetView' requires template arguments
        result = (INDI::WidgetView< INumber > *)new INDI::WidgetView< INumber >((INDI::WidgetView &&)*arg1);
                                                                                       ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:41035:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:41049:34: error: use of class template 'INDI::WidgetView' requires template arguments
  arg1 = reinterpret_cast< INDI::WidgetView * >(argp1);
                                 ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:41054:88: error: use of class template 'INDI::WidgetView' requires template arguments
        result = (INDI::WidgetView< ISwitch > *)new INDI::WidgetView< ISwitch >((INDI::WidgetView const &)*arg1);
                                                                                       ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:41070:9: error: use of class template 'INDI::WidgetView' requires template arguments
  INDI::WidgetView *arg1 = 0 ;
        ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
indiclientpython_wrap.cpp:41073:26: error: use of class template 'INDI::WidgetView' requires template arguments
  std::unique_ptr< INDI::WidgetView > rvrdeleter1 ;
                         ^
/usr/local/include/libindi/indipropertyview.h:348:8: note: template is declared here
struct WidgetView;
       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
8 warnings and 20 errors generated.
error: command '/usr/bin/clang' failed with exit code 1
```

aaronwmorris, 2024-06-03:
This is likely related to https://github.com/indilib/indi/pull/2063 which is included in indi 2.0.8

knro, 2025-05-04:
We released an update version, please check.