indilib/pyindi-client #52: RPi 0 Fails to install

SteveBz, 2025-02-06:
I think I've followied the instructions.  I've installed indi-bin successfully on my RPi zero running 64 bit Bookworm.  I'm trying to use pyindi-client to collect data from indi, by the installation is crashing using `pip install pyindi-client` Here's the error and some truncated output"

'(venv) pi@raspberrypi:~/MoonRabbit $ pip install pyindi-client
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Using cached pyindi-client-1.9.1.tar.gz (18 kB)
  Preparing metadata (setup.py) ... done
Installing collected packages: pyindi-client
  DEPRECATION: pyindi-client is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for pyindi-client ... error
  error: subprocess-exited-with-error

  × Running setup.py install for pyindi-client did not run successfully.
  │ exit code: 1
  ╰─> [861 lines of output]
      /home/pi/MoonRabbit/venv/lib/python3.11/site-packages/setuptools/dist.py:534: UserWarning: Normalizing 'v1.9.1' to '1.9.1'
        warnings.warn(tmpl.format(**locals()))
      running install
      /home/pi/MoonRabbit/venv/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build_ext
      building '_PyIndi' extension
      swigging indiclientpython.i to indiclientpython_wrap.cpp
      swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
      Language subdirectory: python
      Search paths:
         ./
         /usr/include/
         /usr/include/libindi/
         /usr/local/include/libindi/
         ./swig_lib/python/
         /usr/share/swig4.0/python/
         ./swig_lib/
         /usr/share/swig4.0/
      Preprocessing...
      Starting language-specific parse...
      /usr/include/libindi/indiwidgettraits.h:38: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:49: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:60: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:69: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:83: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
.
.
.
....{lots of lines here} ...
.
.
.
      /usr/include/libindi/indiproperties.h:97: Warning 503: Can't wrap 'operator Properties*' unless renamed to a valid identifier.
      /usr/include/libindi/indiproperties.h:98: Warning 503: Can't wrap 'operator const Properties*' unless renamed to a valid identifier.
      Processing types...
      C++ analysis...
      Processing nested classes...
      Generating wrappers...
      error: command '/usr/bin/swig' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pyindi-client

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

'

knro, 2025-05-04:
We release an updated version, can you please test again? and use --no-cache option with pip

SteveBz, 2025-05-24:
OK, successful:

> sudo apt-get update
> sudo apt-get install libdbus-1-dev pkg-config cmake
> sudo apt-get install libglib2.0-dev
> pip install pyindi-client --no-cache-dir

Successfully built dbus-python
Installing collected packages: bottle, dbus-python, pyindi-client
Successfully installed bottle-0.13.3 dbus-python-1.4.0 pyindi-client-2.1.9

Thanks.

Steve.