indilib/pyindi-client #54: PyIndi.py module not being installed

aaronwmorris, 2025-02-07:
In the recent updates, it appears that the build process is not installing the `PyIndi.py` module.  I tried to fix this in #53, which works when installing locally, but not when building directly from git.

What is tricky about this is if you already have the module installed, you will not notice the problem since the file already exists.  You have to fully uninstall pyindi-client before re-installing to see the failure.

SteveBz, 2025-02-08:
For me, on RPi 0, and Bookworm, it doesn't install at all.  Not with `pip install pyindi-client` or with `pip install pyindi-client --no-cache-dir`, even after Jasem's recent update.

I get:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Downloading pyindi-client-1.9.1.tar.gz (18 kB)
  Preparing metadata (setup.py) ... done
Installing collected packages: pyindi-client
  DEPRECATION: pyindi-client is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
```

Is that the same as your error?

Steve.

knro, 2025-02-08:
The new pyindi-client uses TOML project file, so this is updated already. The problem is that the CI/CD job fails. If it is successfully, it should upload the latest version to pypi as the one there is outdated. 

I need help on this as I exhausted several hours trying to diagnose the docker build issue. I even asked a devops engineer to look at this and it's still not resolved.

SteveBz, 2025-02-08:
I'm debugging at the moment with chatgpt. I'll let you know.

Steve.

⁣Get BlueMail for Android ​

aaronwmorris, 2025-02-21:
@tbowers7 While we figure this out the problem with the new build process, install using the older build.  There is no difference with the indi integrations.

```
pip install "git+https://github.com/indilib/pyindi-client.git@d8ad88f#egg=pyindi-client"
```

coutanto, 2025-03-14:
I had the same problem as tbowers7 when installing on raspberry 5. I solved it by changing the line:
sources = ["PyIndi.py", "indiclientpython.i"]
into
sources = ["indiclientpython.i"]
in pyproject.toml

I also changed the swig options into  pyproject.toml and setup.cfg: remove -thread and add -python
(swig version 4.1.0, virtual env with miniconda and python 3.11)

aaronwmorris, 2025-03-14:
@coutanto Adding `PyIndi.py` to the sources was my attempt to fix the real underlying problem.  If you fully uninstall pyindi-client and try the build, the PyIndi.py file is never installed.  It worked for you because the PyIndi.py file was already in place from a previous install.

aaronwmorris, 2026-09-23:
Bumping this issue.  PyIndi.py is still not being installed as of the latest commit a1f1b3a

knro, 2026-09-23:
@s-pouya-sh Please investigate

tbowers7, 2026-09-23:
Found the problem, and submitted PR #68 to finally fix this issue.

closed 2026-09-24 by knro