indilib/pyindi-client #57: Documentation of AbstractBaseClient methods such as connectDevice,disconnectDevice,sendNewNumber,sendNewSwitch...

gnthibault, 2025-04-23:
To whom it may concern

While checking the documentation, I realized that there were some ambiguity as to how the client and device objects are supposed to be used. For instance, I see that the following BaseDevice method is not connected:
* isConnected

For AbstractBaseClient there are many more undocumented methods:
* connectDevice 
* disconnectDevice
* sendNewNumber
* sendNewSwitch
* [...]

Can someone confirm wether the connectDevice is equivalent to setting the "CONNECTION" switch vector element "CONNECT" to On ? (respectively DISCONNECT for AbstractBaseClient.disconnectDevice) ?

Also, with regard to the actual operating model, may we have some additional elements, such as, which call does send an actual update to the server ?
It is ambiguous for me wether it is supposed to be, for instance
switchProperty[0].setState(PyIndi.ISS_ON)
or rather
indiClient.sendNewSwitch(switchProperty)

Thank you in advance for your help. Any pointer to source C++ code with basic explanation is welcome.

knro, 2025-04-24:
Did you check the [INDI Client](https://docs.indilib.org/clients/) documentation?

gnthibault, 2025-04-24:
Indeed it provides a valuable insight, I will try to answer my own questions here, also, I noticed valuable updated sample here that I had missed before: https://github.com/indilib/pyindi-client/blob/cc261891abf0bb1c835fa6681187f251b4ef870e/examples/tuto-scripts/tuto-ccd.py

gnthibault, 2025-04-24:
Ok I am sorry I cannot contribute more at the moment regarding documentation, howver, if anyone comes across this, I have done a small (and somewhat clumsy) test to check behaviour of some of the client features:

```
# Generic

# PyIndi, for a better undesrtanding of client model, see https://docs.indilib.org/clients/
import PyIndi

# for logging
import sys
import time
import logging

# import the PyIndi module
import PyIndi

logging.basicConfig(format='%(asctime)s %(message)s', level=logging.INFO)
logger = logging.getLogger(__name__)

# class IndiClient(PyIndi.BaseClient):
#     def __init__(self):
#         super(IndiClient, self).__init__()

# The IndiClient class which inherits from the module PyIndi.BaseClient class
# Note that all INDI constants are accessible from the module as PyIndi.CONSTANTNAME
class IndiClient(PyIndi.BaseClient):
    def __init__(self):
        super(IndiClient, self).__init__()
        logger.info('creating an instance of IndiClient')

    def newDevice(self, d):
        '''Emmited when a new device is created from INDI server.'''
        logger.info(f"new device {d.getDeviceName()}")

    def removeDevice(self, d):
        '''Emmited when a device is deleted from INDI server.'''
        logger.info(f"remove device {d.getDeviceName()}")

    def newProperty(self, p):
        '''Emmited when a new property is created for an INDI driver.'''
        logger.info(f"new property {p.getName()} as {p.getTypeAsString()} for device {p.getDeviceName()}")

    def updateProperty(self, p):
        '''Emmited when a new property value arrives from INDI server.'''
        logger.info(f"update property {p.getName()} as {p.getTypeAsString()} for device {p.getDeviceName()}")

    def removeProperty(self, p):
        '''Emmited when a property is deleted for an INDI driver.'''
        logger.info(f"remove property {p.getName()} as {p.getTypeAsString()} for device {p.getDeviceName()}")

    def newMessage(self, d, m):
        '''Emmited when a new message arrives from INDI server.'''
        logger.info(f"new Message {d.messageQueue(m)}")

    def serverConnected(self):
        '''Emmited when the server is connected.'''
        logger.info(f"Server connected ({self.getHost()}:{self.getPort()})")

    def serverDisconnected(self, code):
        '''Emmited when the server gets disconnected.'''
        logger.info(f"Server disconnected (exit code = {code},{self.getHost()}:{self.getPort()})")

def test_PyIndiClientConnection():
    # Create an instance of the IndiClient class and initialize its host/port members
    indiclient = IndiClient()
    indiclient.setServer("localhost", 7624)
    assert indiclient.getHost() == "localhost"
    assert indiclient.getPort() == 7624
    res = indiclient.connectServer()
    assert res
[... 276 lines cut ...]
#  'getLight',
#  'getMediator',
#  'getNumber',
#  'getProperties',
#  'getProperty',
#  'getPropertyPermission',
#  'getPropertyState',
#  'getRawProperty',
#  'getSharedFilePath',
#  'getSwitch',
#  'getText',
#  'isConnected',
#  'isDeviceNameMatch',
#  'isValid',
#  'lastMessage',
#  'messageQueue',
#  'registerProperty',
#  'removeProperty',
#  'setDeviceName',
#  'setMediator',
#  'setValue',
#  'this',
#  'thisown',
#  'watchProperty']
```

closed 2025-04-24 by gnthibault