indilib/pyindi-client #59: Segmentation fault in PyIndi

gnthibault, 2025-04-26:
To whom it may concern,

I am not sure how to debug this further, but it definitely seems that something is wrong with the way I am using PyIndi client.
I am randomly experiencing segmentation fault in python...

I enabled, fault handler with:
`export PYTHONFAULTHANDLER=1
`
The last relevant I got ended with this:

> Current thread 0x0000f90b2056b020 (most recent call first):
  File "/opt/venv/lib/python3.12/site-packages/PyIndi/PyIndi.py", line 722 in sendNewSwitch

Unfortunately, I am not sure where to go from there...

knro, 2025-04-27:
Do you have a sample script to show this issue?

gnthibault, 2025-04-27:
Unfortunately no, the python fault handler is actually reporting  10 thread callstack, the app behind is somewhat a bit complex, but most importantly, the issue appears at random, so I cannot really tell what exactly is triggering the issue. I have now huge suspissions about the callback mechanism from PyIndi client virtual method override (in python). 
Do you have any advise about how to write those / what to be careful about so that I can maybe focus on certain parts of the code ?
I'll try to do a few run with strace to try to see if I can get more insights.
Thank you in advance for your help.

gnthibault, 2025-04-27:
Unfortunately using strace did not brought me a lot more informations:

> write(28, "<newSwitchVector device='", 25) = 25
> --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
> rt_sigaction(SIGSEGV, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, NULL, 8) = 0
> write(2, "Fatal Python error: ", 20Fatal Python error: )    = 20
> write(2, "Segmentation fault", 18Segmentation fault)      = 18
> 

I have really no Idea how writing a xml string to a socket file descriptor (which I think is what is happening in the C++ client) could result in a segfault...
The worse is that this issue arise randomly...

gnthibault, 2025-04-27:
Ah, one question that came to my mind, is it possible to trigger a "getProperties" request from the PyIndi client by running a connectServer ?

gnthibault, 2025-04-28:
Potentially linked: https://github.com/swig/swig/issues/3086

gnthibault, 2025-04-28:
Additional elements after running GDB.

> Thread 1 "python3.12" received signal SIGSEGV, Segmentation fault.
> --Type <RET> for more, q to quit, c to continue without paging--bt
> userio_xml_escape (io=io@entry=0x7fffdd551750 <INDI::AbstractBaseClientPrivate::io>, user=user@entry=0x55555796c110, src=src@entry=0x0)
>     at /home/gnthibault/projects/indi/libs/indicore/userio.c:81
> 81	    for(; *ptr; ++ptr)
> (gdb) bt
> #0  userio_xml_escape (io=io@entry=0x7fffdd551750 <INDI::AbstractBaseClientPrivate::io>, user=user@entry=0x55555796c110, src=src@entry=0x0)
>     at /home/gnthibault/projects/indi/libs/indicore/userio.c:81
> #1  0x00007fffdd461867 in IUUserIONewSwitch (io=io@entry=0x7fffdd551750 <INDI::AbstractBaseClientPrivate::io>, user=user@entry=0x55555796c110, svp=0x0)
>     at /home/gnthibault/projects/indi/libs/indicore/indiuserio.c:243
> #2  0x00007fffdd484978 in INDI::AbstractBaseClient::sendNewSwitch (this=this@entry=0x5555575c2680, pp=...)
>     at /home/gnthibault/projects/indi/libs/indidevice/property/indipropertyview.h:368
> #3  0x00007fffdd41fde4 in _wrap_AbstractBaseClient_sendNewSwitch__SWIG_0 (args=<optimized out>) at /usr/include/c++/9/ext/atomicity.h:96
> #4  _wrap_AbstractBaseClient_sendNewSwitch (self=<optimized out>, args=<optimized out>) at indiclientpython_wrap.cpp:20728
> #5  0x00007ffff7abf3e7 in cfunction_call (func=<built-in method AbstractBaseClient_sendNewSwitch of module object at remote 0x7fffdd569170>, 
>     args=(<IndiClient(__version__='0.0.0', config={'name': 'Remote observatory', 'observatory': {'module': 'Observatory', 'investigator': 'gnthibault', 'latitude': <float at remote 0x7fffdce2d8b0>, 'longitude': <float at remote 0x7fffdd586630>, 'elevation': <float at remote 0x7fffdce2d890>, 'timezone': 'Europe/Paris', 'utc_offset': 1, 'horizon': {0: 0, 90: 0, 180: 0, 270: 0}, 'twilight_horizon': -18, 'gmt_offset': 60, 'scope_controller': {'module': 'DummyScopeController', 'port': '/dev/ttyACM0', 'controller_name': 'Arduino', 'indi_driver_connect_delay_s': 5, 'indi_camera_driver_name': 'Canon DSLR', 'indi_mount_driver_name': 'Losmandy Gemini', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}, 'dome_controller': {'module': 'IndiDomeController', 'dome_movement_timeout_s': 300, 'dome_name': 'Dome Simulator', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}}, 'directories': {'base': '/opt/RemoteObservatory', 'images': '/opt/RemoteObservatory/im...(truncated), kwargs=0x0) at Objects/methodobject.c:548
> #6  0x00007ffff7a65ed8 in _PyObject_Call (kwargs=0x0, 
>     args=(<IndiClient(__version__='0.0.0', config={'name': 'Remote observatory', 'observatory': {'module': 'Observatory', 'investigator': 'gnthibault', 'latitude': <float at remote 0x7fffdce2d8b0>, 'longitude': <float at remote 0x7fffdd586630>, 'elevation': <float at remote 0x7fffdce2d890>, 'timezone': 'Europe/Paris', 'utc_offset': 1, 'horizon': {0: 0, 90: 0, 180: 0, 270: 0}, 'twilight_horizon': -18, 'gmt_offset': 60, 'scope_controller': {'module': 'DummyScopeController', 'port': '/dev/ttyACM0', 'controller_name': 'Arduino', 'indi_driver_connect_delay_s': 5, 'indi_camera_driver_name': 'Canon DSLR', 'indi_mount_driver_name': 'Losmandy Gemini', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}, 'dome_controller': {'module': 'IndiDomeController', 'dome_movement_timeout_s': 300, 'dome_name': 'Dome Simulator', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}}, 'directories': {'base': '/opt/RemoteObservatory', 'images': '/opt/RemoteObservatory/im...(truncated), callable=<built-in method AbstractBaseClient_sendNewSwitch of module object at remote 0x7fffdd569170>, 
>     tstate=0x7ffff7fc5108 <_PyRuntime+459656>) at Objects/call.c:367
> #7  _PyObject_Call (tstate=0x7ffff7fc5108 <_PyRuntime+459656>, callable=<built-in method AbstractBaseClient_sendNewSwitch of module object at remote 0x7fffdd569170>, 
>     args=(<IndiClient(__version__='0.0.0', config={'name': 'Remote observatory', 'observatory': {'module': 'Observatory', 'investigator': 'gnthibault', 'latitude': <float at remote 0x7fffdce2d8b0>, 'longitude': <float at remote 0x7fffdd586630>, 'elevation': <float at remote 0x7fffdce2d890>, 'timezone': 'Europe/Paris', 'utc_offset': 1, 'horizon': {0: 0, 90: 0, 180: 0, 270: 0}, 'twilight_horizon': -18, 'gmt_offset': 60, 'scope_controller': {'module': 'DummyScopeController', 'port': '/dev/ttyACM0', 'controller_name': 'Arduino', 'indi_driver_connect_delay_s': 5, 'indi_camera_driver_name': 'Canon DSLR', 'indi_mount_driver_name': 'Losmandy Gemini', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}, 'dome_controller': {'module': 'IndiDomeController', 'dome_movement_timeout_s': 300, 'dome_name': 'Dome Simulator', 'indi_client': {'indi_host': 'localhost', 'indi_port': 7624, 'use_unique_client': True}}}, 'directories': {'base': '/opt/RemoteObservatory', 'images': '/opt/RemoteObservatory/im...(truncated), kwargs=0x0) at Objects/call.c:339
> #8  0x00007ffff79fd767 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=0x7ffff78c5a70, throwflag=<optimized out>) at Python/bytecodes.c:3254
> #9  0x00007ffff7a67693 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=2, args=0x7fffffffd1e0, callable=<function at remote 0x7fffe0ae6b60>, 
>     tstate=0x7ffff7fc5108 <_PyRuntime+459656>) at ./Include/internal/pycore_call.h:92
[... cut ...]
> #18 0x00007ffff7c7ab09 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>, args=<optimized out>, callable=<method at remote 0x7fffdc1906c0>, 
>     tstate=0x7ffff7fc5108 <_PyRuntime+459656>) at ./Include/internal/pycore_call.h:92
> #19 partial_vectorcall (pto=<optimized out>, args=0x7ffff78c51e8, nargsf=9223372036854775808, kwnames=0x0) at ./Modules/_functoolsmodule.c:230
> #20 0x00007ffff7a63fe5 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=9223372036854775808, args=0x7ffff78c51e8, 
>

gnthibault, 2025-04-28:
Ok I think I found the issue. The rootcause was pretty simple, but eventually pretty hard to find-out...
While re-writing my indi client class, I did a mistake, and I was "simply" retrieving a switch with the following code:

In this case, what happened is that I was reusing a wrong switchVector (isValidI() == False) to be resend through client... And ended up with the aforementionned issue.

I should have used instead a retry logic like in this sample:
https://github.com/indilib/pyindi-client/blob/cc261891abf0bb1c835fa6681187f251b4ef870e/examples/pyindi-synscan.py#L78

I will fix that, and everything should be alright. Will report findings, and eventually close the issue in the upcoming days