indilib/pyindi-client #61: getblobdata() leads to segmentation fult in python with pyindi client.

jmalgarin86, 2025-08-31:
Hi,

I’m new here — my name is José Miguel. I’m developing a custom GUI with PyIndi to control my old telescope. The setup consists of a mount driven by two stepper motors (controlled via Arduino) and two cameras: a guiding camera (for guiding corrections) and a main imaging camera.

The issue I’m facing is that when I use the getblobdata() method on a BLOB in PyIndi, I occasionally encounter segmentation faults that crash the entire Python process (no Python traceback, just Segmentation fault).

My environment:

Python 3.11

INDI server running with ZWO cameras

Most of the time it works fine, but sometimes the crash occurs right after the exposure completes, when reading getblobdata().

I will be happy to prepare a standalone script if you wish.

With faulthandler enabled, I see the following fatal Python error:

```bash
Segmentation fault Current thread 0x00007fff15acf180 (most recent call first): <no Python frame> Thread 0x00007fff31f34700 (most recent call first): File "/home/astropi/TelescopeControllerSourceCode/.venv/lib/python3.11/site-packages/PyIndi.py", line 575 in getblobdata File "/home/astropi/TelescopeControllerSourceCode/controllers/controller_camera.py", line 389 in capture File "/home/astropi/TelescopeControllerSourceCode/controllers/controller_camera.py", line 586 in <module> Extension modules: _PyIndi, numpy._core._multiarray_umath, numpy.linalg._umath_linalg, PyQt5.QtCore, PyQt5.QtGui, PyQt5.QtWidgets, astropy.io.fits._utils, astropy.io.fits.hdu.compressed._compression, numcodecs.compat_ext, numcodecs.blosc, numcodecs.zstd, numcodecs.lz4, numcodecs._shuffle, numcodecs.jenkins, crc32c._crc32c, numcodecs.vlen, numcodecs.fletcher32, PIL._imaging, kiwisolver._cext (total: 19)
```

knro, 2025-09-01:
Thank you for the report. Please create a minimal example to reproduce the crash when used with CCD Simulator

jmalgarin86, 2025-09-02:
Hi,

I prepared a small example. After run the example I could reproduce the error with both my ZWO camera and the simulator. You can get the example here:

https://github.com/jmalgarin86/TelescopeController/blob/dev/controllers/controller_camera_standalone.py

Thanks